HelTecAutomation/ESP32_LoRaWAN #1: Wrong pin for DIO1

michaelshmitty, 2018-12-07:
https://github.com/HelTecAutomation/ESP32_LoRaWAN/blob/b34e3eea56b09c126cab389d23016f01ebcda916/examples/OTAA/OTAA.ino#L44

The hardware DIO1 pin of the SX1276 is connected to pin 25 of the HelTec board, not pin 35 as stated in the examples.

Heltec-Aaron-Lee, 2018-12-21:
the new version WiFi LoRa 32 had a different pin out.
V1 pin out diagram: [](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32_V1.pdf)
V2 pin out diagram: [](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32_V2.pdf)

Heltec-Aaron-Lee, 2018-12-21:
we have two version WiFi LoRa 32 baords, they had a different pin defintion on DIO1.
V1 pin out diagram: [https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32_V1.pdf](url)
V2 pin out diagram: [https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32_V2.pdf](url)