HelTecAutomation/ESP32_LoRaWAN #12: Can't join after a while E868

OstertagM, 2019-08-29:
Dear Support,
I have tested with the example when joins work for some times.
But after some test i didn't any more! 
But I see on my gateway the the join is accepted but it will not be received!
What happens?
[image]

DeveloppeurPascal, 2019-09-10:
Same problem in France with Orange LoRa Network.

solotaker, 2019-09-16:
This looks like the node missed the wake up time,
Are you using example OTAA_OLED?

OstertagM, 2019-09-16:
Yes I used this.

solotaker, 2019-09-17:
I have a suggestion, have you tried to test the CLASS C scene, because in C mode, the node will not enter sleep mode and keep the RX2 window open, which will ensure that the downstream frame will be used during the effective time of the node. Whether the node has a defect on the hardware.

OstertagM, 2019-09-17:
Yeah, try and error game?
I want to build up as fast as possible an low power node. At first with oled for low power without. But heaving same code base. 
Best I go with ADB and lmic perhaps. 
@solotaker or do you have an good ready working code to post?

solotaker, 2019-09-18:
Hi:
I am wondering why your LoRa_32 can't successfully enter the network?
My LORA_32 868 frequency has been working on TTN for a long time.
I am using the example "OTAA_OLED".
[image]

OstertagM, 2019-09-18:
I work's for about 20 joins during testing. Suddenly not. I have the new TTN Gateway.
I have decided that OTAA is something if you you know you will get the downlink each time during the join process. I also have decided to go with lmic because that has worked with ADB at once. 
Also seasons saving is working here good. Because in case of deep sleep a new join is consumed a lot of to much power. 
And with lmic I can receive downlink I tested it with my own data.

DeveloppeurPascal, 2019-09-18:
Hi Mathias, I have a problem with OTAA too (and I suppose there is a conflict with SD Card libraries, I’ll confirm it after some tests).
Can you tell me what lmic library you choosed ? There is a lot on there.

OstertagM, 2019-09-18:
Hi Pascal. I use https://github.com/mcci-catena/arduino-lmic and in abp mode.
You can also read this topic https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-3/18436/247.
Hopefully from your response

DeveloppeurPascal, 2019-09-18:
Thanks, I tried this library before but didn't use it successfully. I will retry.