HelTecAutomation/ESP32_LoRaWAN #13: Compilation Error with OTAA_DHT11

NoBiXXL, 2019-08-30:
Trying to compile OTAA_DHT11.ino (from examples) with Arduino IDE.
Compilation stopped with the message below.
I have been successful with OTAA_OLED which should use the same principles!
I have noticed, that in OTAA_OLED CLASS is defined (which is not in OTAA_DHT11)!
Also the call for LoRa.DeviceSleep in OTAA_OLED is different to OTAA_DHT11!
What is wrong here?

D:\HomeOrdnerNH\3D\Arduion IDE\Sketches\OTAA_DHT11\OTAA_DHT11.ino: In function 'void loop()':

OTAA_DHT11:182:28: error: no matching function for call to 'LoRaClass::DeviceStateInit()'

       LoRa.DeviceStateInit();

                            ^

In file included from D:\...\Arduion IDE\Sketches\OTAA_DHT11\OTAA_DHT11.ino:47:0:

D:\....\Arduion IDE\Sketches\libraries\ESP32_LoRaWAN-master\src/LoRa.h:87:8: note: candidate: void LoRaClass::DeviceStateInit(DeviceClass_t)

   void DeviceStateInit(DeviceClass_t CLASS);

        ^

D:\....\Arduion IDE\Sketches\libraries\ESP32_LoRaWAN-master\src/LoRa.h:87:8: note:   candidate expects 1 argument, 0 provided

OTAA_DHT11:227:24: error: 'IsLowPowerOn' was not declared in this scope

       LoRa.DeviceSleep(IsLowPowerOn,DebugLevel);

                                   ^

exit status 1
no matching function for call to 'LoRaClass::DeviceStateInit()'

solotaker, 2019-09-16:
Hi:
I have fixed these issues.
Sorry, the DHT11 example has been in disrepair for a long time.

NoBiXXL, 2019-09-16:
Thanksgiving! 
I have seen it and already on the way to integrate the new sections in my project. 
After a successful test the issue will be closed. 
Best regards Norbert 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/HelTecAutomation/ESP32_LoRaWAN/issues/13?email_source=notifications\u0026email_token=AM52CF4DQLDERI3B4LWANWDQJ5PUPA5CNFSM4ISPC6JKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD6YX6PQ#issuecomment-531726142",
"url": "https://github.com/HelTecAutomation/ESP32_LoRaWAN/issues/13?email_source=notifications\u0026email_token=AM52CF4DQLDERI3B4LWANWDQJ5PUPA5CNFSM4ISPC6JKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD6YX6PQ#issuecomment-531726142",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

NoBiXXL, 2019-09-16:
I have now tried to get the code alive...and failed again.
I think that some few changes need to be done in addition:
1) Define in OTTA_DHT11.ino in the definition section the class: #define  CLASS  CLASS_A 
2) Change in the loop part of the sketch: LoRa.DeviceStateInit() -> LoRa.DeviceStateInit(CLASS)
3) Change in the loop part of the sketch: LoRa.DeviceSleep(IsLowPowerOn,DebugLevel) ->LoRa.DeviceSleep(CLASS,DebugLevel);

After these changes the sketch compiles and runs on the ESP32 HELTEC V2

solotaker, 2019-09-17:
Yes, that's true, these operations have been done in the latest version of the library.

closed 2019-09-17 by NoBiXXL