HelTecAutomation/ESP32_LoRaWAN #16: Changing APP EUI, KEY and DEV EUI without changing defined in the library

DeveloppeurPascal, 2019-09-10:
An idea for a next release of this library.

It is convenient to have OTAA parameters (APP_EUI, APPKEY, DEV_EUI) as defined in the library when you only have one device, but when you have several devices it is useless. It must be possible to change them as a variable in the code of our programs.

As it stands, the proposed library does not allow to industrialize the work with Heltec cards. That's a pity.

solotaker, 2019-09-16:
This is very easy to do, thank you for your suggestion, maybe we will make adjustments in the next version.

DeveloppeurPascal, 2019-09-16:
Thanks. 
For the others I fixed the problem for me by changing values to DevEui[], AppEui[] and AppKey[] global variables as first lines of setup() function.

The last great limit is for the LICENSE variable and MCU activation.

DeveloppeurPascal, 2019-09-16:
For automating DevEui, I change it from ChipId with this code :

  uint64_t chipid = ESP.getEfuseMac();
  for (int i = 0; i < 8; i++) {
    if (i < 2) {
      DevEui[i] = 0;
    }
    else {
      DevEui[i] = chipid % 256;
      chipid = chipid / 256;
    }
  }

So if ChipId is 12 34 56 78 9A BC, my DevEui will be 00 00 BC 9A 78 56 34 12.

DeveloppeurPascal, 2019-09-17:
My code works, but the library don't use the new value of DevEui[], only the Defines in commissioning.h :-(

Ajaynithish, 2019-12-21:
@DeveloppeurPascal  where to mention the user data in this library? i can't find one
`void PrepareMsgFrame( uint8_t port, uint8_t Msg2Send[], uint8_t length ) ` is this is the one? @solotaker @Heltec-Aaron-Lee @gibus27

solotaker, 2019-12-23:
> @DeveloppeurPascal where to mention the user data in this library? i can't find one
> `void PrepareMsgFrame( uint8_t port, uint8_t Msg2Send[], uint8_t length ) ` is this is the one? @solotaker @Heltec-Aaron-Lee @gibus27

Yes, use like this：
```
      uint8_t msg[] = "Msg test 43";
      PrepareMsgFrame( AppPort, msg , sizeof(msg)-1);
```

Ajaynithish, 2019-12-30:
@DeveloppeurPascal  how did you defined the parameters outside the library. I've been trying it for weeks, all I get is ERROR:undefined reference. any one any hint @solotaker @gibus27 @Heltec-Aaron-Lee

Heltec-Aaron-Lee, 2020-01-03:
what parameters?

Ajaynithish, 2020-01-03:
The dev addr , netsessionkey.... But i found it thanks for the response