HelTecAutomation/ESP32_LoRaWAN #19: EU868 Compile Issue

patmolloy, 2019-10-12:
I am trying to use OTAA_DHT11.ino on the Heltec Wireless Stick

In LoRaMac-definitions.h is the line ..

`#define  USE_BAND_915`

If I compile everything (in the arduino compiler) it all works, and loads to the board correctly -- but of course on the *wrong* frequency for my region (915).

So I change this line to

`#define  USE_BAND_868`

and re-compile, I get the following error ...

```
C:\Users\patmo\OneDrive\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\LoRaMac.c: In function 'SetNextChannel':

C:\Users\patmo\OneDrive\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\LoRaMac.c:2038:60: error: expected expression before '==' token

                         if( ( JOIN_CHANNELS & ( 1 << j ) ) == 0 )

                                                            ^

C:\Users\patmo\OneDrive\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\LoRaMac.c:2038:65: error: expected statement before ')' token

                         if( ( JOIN_CHANNELS & ( 1 << j ) ) == 0 )

                                                                 ^
```

Can you help?

patmolloy, 2019-10-12:
I have resolved this by including the #defines here https://stackforce.github.io/LoRaMac-doc/group___r_e_g_i_o_n_e_u868.html and changing JOIN_CHANNELS to EU868_JOIN_CHANNELS 

This works for EU868 but is NOT a general solution

legapi, 2019-10-12:
> I am trying to use OTAA_DHT11.ino on the Heltec Wireless Stick
>
> In LoRaMac-definitions.h is the line ..
>
> `#define USE_BAND_915`
>
> If I compile everything (in the arduino compiler) it all works, and [...]

This error comes from the definition of "**JOIN_CHANNELS**" at line **490** of **LoRaMac-definition.h**.
If you compare the 868MHz frequency to 433MHz and 780MHz frequency, you may see more parentheses.

868MHz (line 490)
`#define JOIN_CHANNELS      ( uint16_t )( LC( 1 ) )| LC( 2 ) | LC( 3 ) )`
...where maybe it should be...
`#define JOIN_CHANNELS      ( uint16_t )( LC( 1 ) | LC( 2 ) | LC( 3 ) )`  <correct>

433MHz (line 142)
`#define JOIN_CHANNELS      ( uint16_t )( LC( 1 ) | LC( 2 ) | LC( 3 ) )`

780MHz (line 361)
`#define JOIN_CHANNELS      ( uint16_t )( LC( 1 ) | LC( 2 ) | LC( 3 ) )`

patmolloy, 2019-10-12:
Brilliant !! Thanks :) That is definitely the issue and your solution is much better than my bodge!