HelTecAutomation/ESP32_LoRaWAN #20: Send packets larger than 11 bytes at 915MHz frequency

legapi, 2019-10-19:
Hi.

I am trying to transmit packets larger than 11 bytes (defined in Commissioning.h line: 136, `LORAWAN_APP_DATA_SIZE`), in addition I changed the AppData [] array as follows:

```
...
             AppData [9] = '0';
             AppData [10] = 'a';
             AppData [11] = 'b';
             AppData [12] = 'c';
             AppData [13] = 'd';
...
```

I just changed the `LORAWAN_APP_DATA_SIZE `to 14 bytes, but according to my gateway the packet is not received.

What must be changed to send 14 bytes in AppData?

andrelmbraga, 2020-01-06:
I have the same problem. I'm working with 915 MHz, therefore I have changed in commissionig.h  the 
LORAWAN_APP_DATA_SIZE to 12 in order to test, but without succeced. 

```
* User application data buffer size
*/
#define LORAWAN_APP_DATA_MAX_SIZE                           64
/*!
 * User application data buffer size
 */
#if defined( USE_BAND_433 ) || defined( USE_BAND_470 ) || defined( USE_BAND_470PREQUEL ) || defined( USE_BAND_780 ) || defined( USE_BAND_868 )
#define LORAWAN_APP_DATA_SIZE                       16
#elif defined( USE_BAND_915 ) || defined( USE_BAND_915_HYBRID )
#define LORAWAN_APP_DATA_SIZE                       11
#endif
```

andrelmbraga, 2020-01-10:
Solution for the problem:

http://community.heltec.cn/t/solved-send-more-than-11-bytes-915-mhz-lorawan/752

I think you can close this issue.

legapi, 2020-04-09:
Tanks.

closed 2020-04-09 by legapi