HelTecAutomation/ESP32_LoRaWAN #24: Add AS923 frequency plan to LoRaMac.c

s0301132, 2019-12-05:
Hi @Heltec-Aaron-Lee @solotaker @gibus27 !

I have some issue when I set the ESP32 V2 to TTN network, my region use AS2 frequency plan (AS923-925), so when I config the `Commissioning.h` file I can't choose the suitable frequency plan. Is it possible to add a frequency plan to `LoRaMac.c` file to add a new frequency plan? Thanks!

I find a RegionAS923 config files from repositories LoRaMac-node see if it can help:
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/region/RegionAS923.h
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/region/RegionAS923.c

Thanks a lot!

Heltec-Aaron-Lee, 2019-12-06:
OK, please give us a few days.

s0301132, 2019-12-06:
@Heltec-Aaron-Lee 

OK Thanks a lot !

shmrymbd, 2019-12-13:
me also need your help of AS923 (AS1). Thanks

Heltec-Aaron-Lee, 2020-02-10:
Hi, there, I did a big update in the last few days, and it's already have full region support. Please try the newest library.

[image]

s0301132, 2020-02-15:
Hi @Heltec-Aaron-Lee !
Thanks a lot! I reinstall the development board environment and this library and now I can join the network successfully!

However, there are some new problems:
1. The node can only send data once and stuck on sending process 
[image]

[image]

I'm using us915 bandwidth and ttn network with my esp32 lora node and HT-M01 LoRa Gateway

s0301132, 2020-02-15:
Hi @Heltec-Aaron-Lee 
Also will you guys update the HT-M01 gateway library with full region support so I can use both as932 frequency on both esp32 board and gateway. Thanks !

s0301132, 2020-02-18:
Hi @Heltec-Aaron-Lee 
I fix the problem when I switch to EU868 frequency plan, I think the problem is TTN use sub band 2 in US 915 but esp32 node use sub band 1 in US915, so the message downlink may have some error,  how can I change the sub band in esp32 just like cubecell ?

Heltec-Aaron-Lee, 2020-02-19:
@s0301132 Hi, I had updated the code last few days many times, please use the "git pull" update again.

About the sub-band usage, please refer to this document, maybe it makes sense to you.
[https://heltec-automation-docs.readthedocs.io/en/latest/general/sub_band_usage.html](https://heltec-automation-docs.readthedocs.io/en/latest/general/sub_band_usage.html)