HelTecAutomation/ESP32_LoRaWAN #25: Cannot build example OTAA_OLED using PlatformIO

torkleyy, 2019-12-11:
I'm trying to build [this example](https://github.com/HelTecAutomation/ESP32_LoRaWAN/tree/master/examples/OTAA_OLED) using PlatformIO with the following config:

```ini
[env:heltec_wifi_lora_32_V2]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = arduino

lib_deps =
    ESP32_LoRaWAN@2.0
```

This is the output of build:

```
Processing heltec_wifi_lora_32_V2 (platform: espressif32; board: heltec_wifi_lora_32_V2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_lora_32_V2.html
PLATFORM: Espressif 32 1.11.1 > Heltec WiFi LoRa 32 (V2)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32_LoRaWAN> 2.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
Building in release mode
Compiling .pio/build/heltec_wifi_lora_32_V2/src/main.cpp.o
Generating partitions .pio/build/heltec_wifi_lora_32_V2/partitions.bin
Compiling .pio/build/heltec_wifi_lora_32_V2/libd88/Wire/Wire.cpp.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib8c3/SPI/SPI.cpp.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/LoRa.cpp.o
Archiving .pio/build/heltec_wifi_lora_32_V2/lib8c3/libSPI.a
Indexing .pio/build/heltec_wifi_lora_32_V2/lib8c3/libSPI.a
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/LoRaMac.c.o
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/board.h:35:0,
                 from src/main.cpp:34:
.pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/sx1276.h:469:13: warning: 'void RxChainCalibration()' declared 'static' but never defined [-Wunused-function]
 static void RxChainCalibration( void );
             ^
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/board.h:35:0,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/LoRaMac.c:21:
.pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/sx1276.h:469:13: warning: 'RxChainCalibration' declared 'static' but never defined [-Wunused-function]
 static void RxChainCalibration( void );
             ^
Archiving .pio/build/heltec_wifi_lora_32_V2/libd88/libWire.a
Indexing .pio/build/heltec_wifi_lora_32_V2/libd88/libWire.a
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/LoRaMacCrypto.c.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/Mcu.S.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/OLEDDisplay.cpp.o
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/board.h:35:0,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/LoRa.h:6,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/LoRa.cpp:1:
.pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/sx1276.h:469:13: warning: 'void RxChainCalibration()' declared 'static' but never defined [-Wunused-function]
 static void RxChainCalibration( void );
             ^
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/OLEDDisplayUi.cpp.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/aes.c.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/board.c.o
Compiling .pio/build/heltec_wifi_lora_32_V2/lib5b9/ESP32_LoRaWAN_ID6633/cmac.c.o
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/board.h:35:0,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/board.c:15:
.pio/libdeps/heltec_wifi_lora_32_V2/ESP32_LoRaWAN_ID6633/src/sx1276.h:469:13: warning: 'RxChainCalibration' declared 'static' but never defined [-Wunused-function]
[... 75 lines cut ...]
Indexing .pio/build/heltec_wifi_lora_32_V2/libFrameworkArduino.a
Linking .pio/build/heltec_wifi_lora_32_V2/firmware.elf
.pio/build/heltec_wifi_lora_32_V2/lib5b9/libESP32_LoRaWAN_ID6633.a(rtc-board.S.o): In function `TimerGetTimerValue':
D:\/D:\61722\Documents\Arduino\hardware\heltec\esp32\libraries\ESP32_LoRaWAN\src\rtc-board.c:120:(.text.TimerGetTimerValue+0x31): dangerous relocation: call8: call target out of range: __udivdi3
.pio/build/heltec_wifi_lora_32_V2/lib5b9/libESP32_LoRaWAN_ID6633.a(rtc-board.S.o): In function `TimerGetElapsedAlarmTime':
D:\/D:\61722\Documents\Arduino\hardware\heltec\esp32\libraries\ESP32_LoRaWAN\src\rtc-board.c:120:(.text.TimerGetElapsedAlarmTime+0x32): dangerous relocation: call8: call target out of range: __udivdi3
.pio/build/heltec_wifi_lora_32_V2/lib5b9/libESP32_LoRaWAN_ID6633.a(rtc-board.S.o): In function `TimerComputeElapsedTime':
D:\/D:\61722\Documents\Arduino\hardware\heltec\esp32\libraries\ESP32_LoRaWAN\src\rtc-board.c:134:(.text.TimerComputeElapsedTime+0x32): dangerous relocation: call8: call target out of range: __udivdi3
collect2: error: ld returned 1 exit status
*** [.pio/build/heltec_wifi_lora_32_V2/firmware.elf] Error 1
==================================================================================== [FAILED] Took 5.99 seconds ====================================================================================
The terminal process terminated with exit code: 1
```

Note the errors:

```
Linking .pio/build/heltec_wifi_lora_32_V2/firmware.elf
.pio/build/heltec_wifi_lora_32_V2/lib5b9/libESP32_LoRaWAN_ID6633.a(rtc-board.S.o): In function `TimerGetTimerValue':
D:\/D:\61722\Documents\Arduino\hardware\heltec\esp32\libraries\ESP32_LoRaWAN\src\rtc-board.c:120:(.text.TimerGetTimerValue+0x31): dangerous relocation: call8: call target out of range: __udivdi3
.pio/build/heltec_wifi_lora_32_V2/lib5b9/libESP32_LoRaWAN_ID6633.a(rtc-board.S.o): In function `TimerGetElapsedAlarmTime':
D:\/D:\61722\Documents\Arduino\hardware\heltec\esp32\libraries\ESP32_LoRaWAN\src\rtc-board.c:120:(.text.TimerGetElapsedAlarmTime+0x32): dangerous relocation: call8: call target out of range: __udivdi3
.pio/build/heltec_wifi_lora_32_V2/lib5b9/libESP32_LoRaWAN_ID6633.a(rtc-board.S.o): In function `TimerComputeElapsedTime':
D:\/D:\61722\Documents\Arduino\hardware\heltec\esp32\libraries\ESP32_LoRaWAN\src\rtc-board.c:134:(.text.TimerComputeElapsedTime+0x32): dangerous relocation: call8: call target out of range: __udivdi3
```

Tonnard, 2019-12-12:
I have the same issue, it does however build perfectly in Ardunio IDE

moisesfa, 2020-01-04:
Same error in all examples when using this library with Platformio.
Any solution?

moisesfa, 2020-01-22:
I just discovered that it works with platform = espressif32@1.9.0 in the file platformio.ini in my heltec_wireless_stick.
Regards.

torkleyy, 2020-06-03:
I ended up using [lmic](https://github.com/matthijskooijman/arduino-lmic) instead. 

[Working demo](https://github.com/cyberman54/ESP32-Paxcounter/issues/411#issuecomment-520987149)

Waldmensch1, 2020-08-26:
The espressif32@1.9.0 workaround works but any Example ends in a Panic loop on "Joining" so the Library is unusable on PIO for whatever reason.