HelTecAutomation/ESP32_LoRaWAN #26: Join procedure looping-OTAA

Ajaynithish, 2019-12-19:
@Heltec-Aaron-Lee  . I am using the OTAA example- Heltec mini gateway- Heltec wireless shell as node. The join request sent by node is accepted in the server(both things network and Chirp stack).
Problem is it is the only thing repeating again and again(join request and accept). The join accept message is not reaching  the node.
 Thanks
    Ajay

Heltec-Aaron-Lee, 2019-12-19:
Which frequency did you use? How many nodes do you have? Can you try with another one?

Ajaynithish, 2019-12-19:
Actually I am sending at every three default frequencies used in India. Only one node. @Heltec-Aaron-Lee

Heltec-Aaron-Lee, 2019-12-19:
Uhh, I think I know the reason, India uses AS923, but the ESP32_LoRaWAN code still don't have that frequency support.

I will add AS923 and AU915 within the next two days.

Ajaynithish, 2019-12-19:
India have IN865. I have altered the EU868 part of code for Indian specifications and the node is sending at 865 frequency and Thanks for the quick replay

onexecute, 2020-02-06:
Make sure gateway is centered on the frequencies as the thing (end device).

Heltec-Aaron-Lee, 2020-02-10:
I did a big update in the last few days, please try the newest update.