HelTecAutomation/ESP32_LoRaWAN #28: Problem with join to TTN US915

lyakh-m, 2019-12-30:
Hello! I use WiFi LoRa 32 (V2). I modified the OTAA_OLED example for the US 915 range (Comissioning.h: LORAWAN_DEVICE_EUI, LORAWAN_APPLICATION_EUI, LORAWAN_APPLICATION_KEY; LoRaMac-definitions.h: USE_BAND_915; and scetch for correct LICENSE). However, I see only periodic registration (activation request) in the TTN network, and I do not observe data (Payload and answer) transmission at all. What could be the problem?
I see requests going through the router like:
{"rxpk": [
  { "tmst":443640556,
    "chan":3,
    "rfch":0,
    "freq":904.500000,
    "stat":1,
    "modu":"LORA",
    "datr":"SF10BW125",
    "codr":"4/5",
    "lsnr":-6.8,
    "rssi":-95,
    "size":12,
    "data":"QJokAiaA6UW/fzxl"}
]}

{"txpk":
  {"imme":false,
   "tmst":444640556,
   "freq":925.1,
   "rfch":0,
   "powe":20,
   "modu":"LORA",
   "datr":"SF10BW500",
   "codr":"4/5",
   "ipol":true,
   "size":22,
   "ncrc":true,
   "data":"YJokAiaKfgADQAIAcQMFAP8BDe/yFQ=="}
}

[image]
[image]

onexecute, 2019-12-30:
I'm having  a similar issue while working with ChirpStack RAK gateway and I use WiFi LoRa 32 (V2).

13:31:30904.1-116-11-CRC_ERRLORA4/5SF7BW125------13:31:22904.7-119-11.3-CRC_ERRLORAOFFSF7BW125------13:29:48903.9-119-12.3-CRC_ERRLORAOFFSF7BW125------13:29:02904.9-117-11.5-CRC_ERRLORAOFFSF7BW125------13:28:48904.5-118-11.5-CRC_ERRLORA4/7SF7BW125------{
    "freq": 904500000,
    "chan": 3,
    "tmst": 4020452387,
    "utmms": 1577730527072,
    "rfch": 0,
    "stat": -1,
    "rssi": -118,
    "size": 104,
    "modu": "LORA",
    "datr": "SF7BW125",
    "codr": "4/7",
    "lsnr": -11.5,
    "data": "GmyffRKdI2s2TCr9iWW1i+PcvQJRd7bnbXG2nBNdILtF/Lm7z1xdz187CK3egjO5W8WUQ2P6q+r8/jk0dTAE1r4NCl+cx97Oldjr/5aCo0o+dAAA6P5pwmCUtQxTo9k+yKhux03YrVE="
}

andrelmbraga, 2020-01-06:
Similar problem here. I can't see the upload on my application with 915 version.

onexecute, 2020-01-06:
I'm using the RAK 7258 and had to change the Frequency Sub-Band see [http://community.heltec.cn/t/solved-wifi-lora-32-v2-ttn-joining-trouble/670](url)
Now the  joins are sent and received by the ChirpStack server. ChirpStack sends a JoinAccept, the RAK transmits but the device doesn't seem to acknowledge it.

Heltec-Aaron-Lee, 2020-01-07:
Maybe this topic helps you:
[http://community.heltec.cn/t/solved-wifi-lora-32-v2-ttn-joining-trouble/670?u=hunter8801](http://community.heltec.cn/t/solved-wifi-lora-32-v2-ttn-joining-trouble/670?u=hunter8801)

Heltec-Aaron-Lee, 2020-02-10:
I had just made a big update for this library, please try the newest library.

BTW.
[image]

lyakh-m, 2020-02-11:
Thanks a lot!!!

Sincerely
Mikhail Liakhovets