HelTecAutomation/ESP32_LoRaWAN #33: Please commit the source code of Mcu.cpp, timer.cpp, and rtc-board.c

biocage, 2020-02-10:
Hello, 

I was trying to understand and modify functionality in the SDK, but found the files Mcu.S, timer.S, and rtc-board.S in src/ seem to be partially-compiled name-mangled versions of BSD and possibly GPL-licensed open source C/C++ input files (based on preliminary Google searches of symbols present in those files). All three of these assembly-language intermediates, at least as-of git commit fb39aae5588c58de6cbb9ac4e3a4def45b8c8a4a, contain GNU artifacts. Please commit the original source code, and please include proper attribution for original and derivative source code. 

Thank you!

Heltec-Aaron-Lee, 2020-04-09:
Sorry for the late reply.

I rewrite that three files for ESP32 chips, I also hope to open these files as soon as possible, but considering that there are so many bad copy version hardware, I just don't want they use this library directly. and the bad copy hardware also didn't make any independence matching and inaccuracy clock, use this library may cause some unknown problem.

Anyway, if you have any problem or something unclear with this library, we can talk here. 😃

jjanderson, 2020-11-27:
Hi, it seems that there are still files missing:

```
Arduino: 1.8.12 (Windows 10), Board: "CubeCell-Board（HTCC-AB01）, REGION_AS923, CLASS_A, OTAA, ON, CONFIRMED, OFF, ON, ACTIVE, None"

In file included from\home\Arduino\libraries\ESP32_LoRaWAN-master\src/ESP32_LoRaWAN.h:8:0,

                 from \home\Arduino\sketch_nov27a\sketch_nov27a.ino:18:

\home\Arduino\libraries\ESP32_LoRaWAN-master\src/Mcu.h:6:21: fatal error: soc/rtc.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board CubeCell-Board（HTCC-AB01）.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Heltec-Aaron-Lee, 2020-11-28:
Please use the [Heltec ESP32 development framework](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series) and choose the correct board.

closed 2020-11-28 by Heltec-Aaron-Lee