HelTecAutomation/ESP32_LoRaWAN #35: BLE LoraWan Bridge Issue

georaura, 2020-03-04:
I need develop a Bluetooth - LoraWan bridge, but when i insert a default  ESP32 Bluetooth library (BluetoothSerial.h or BLEDevice.h), I have compilation conflicts with ESP32_LoRaWAN.h library. Any suggest?

kkalbaugh, 2020-03-04:
Conflicts where?  Can you post the compile results?

georaura, 2020-03-04:
The problem is with aes.c Library:
The compiled results is: 

------------------------------
libraries\ESP32_LoRaWAN-master\aes.c.o:C:\Users\graura\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src/aes.c:514: first defined here

C:\Users\graura\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.4/tools/sdk/lib\libbt.a(aes.o): In function `aes_cbc_encrypt’:

aes.c:(.text.aes_cbc_encrypt+0x0): multiple definition of `aes_cbc_encrypt’

libraries\ESP32_LoRaWAN-master\aes.c.o:aes.c:(.text.aes_cbc_encrypt+0x0): first defined here 
collect2.exe: error: ld returned 1 exit status
-----------------------------

Conflicting functions are: (aes_set_key and aes_cbc_encrypt).

georaura, 2020-03-04:
To reproduce de error, try to include these libraries to your .ino code:

#include <ESP32_LoRaWAN.h>
#include "Arduino.h"
#include <BLEDevice.h>

justind000, 2020-09-15:
Seeing this also.