HelTecAutomation/ESP32_LoRaWAN #40: problems with sending data to TTN

PTJ2020, 2020-03-21:
Hi, I`d like to send some temperaur datas to ttn. I use the OTAA_OLED example, in which I tried to add a Bmp085 sensor. I dont have much experiance in programming, so can any one help me, to send some datas to ttn?
Thanks for your help.

```
/*
 * HelTec Automation(TM) LoRaWAN 1.0.2 OTAA example use OTAA, CLASS A
 *
 * Function summary:
 *
 * - use internal RTC(150KHz);
 *
 * - Include stop mode and deep sleep mode;
 *
 * - 15S data send cycle;
 *
 * - Informations output via serial(115200);
 *
 * - Only ESP32 + LoRa series boards can use this library, need a license
 *   to make the code run(check you license here: http://www.heltec.cn/search );
 *
 * You can change some definition in "Commissioning.h" and "LoRaMac-definitions.h"
 *
 * HelTec AutoMation, Chengdu, China.
 * 成都惠利特自动化科技有限公司
 * https://heltec.org
 * support@heltec.cn
 *
 *this project also release in GitHub:
 *https://github.com/HelTecAutomation/ESP32_LoRaWAN
*/

#include <ESP32_LoRaWAN.h>
#include "Arduino.h"
#include <Wire.h>
#include <Adafruit_BMP085.h>

Adafruit_BMP085 bmp;

/*license for Heltec ESP32 LoRaWan, quary your ChipID relevant license: http://resource.heltec.cn/search */
uint32_t  license[4] = {0x0234E114,0xBB5A079B,0xEE6F63BD,0x13553A17};
/* OTAA para*/
uint8_t DevEui[] = { 0x00, 0x2F, 0x7C, 0xBF, 0x0E, 0xC5, 0x5D, 0x06 };
uint8_t AppEui[] = { 0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x02, 0xB4, 0xD7 };
uint8_t AppKey[] = { 0x4F, 0x37, 0xCA, 0x11, 0xB6, 0xC5, 0x55, 0x56, 0xB5, 0x88, 0x8A, 0x63, 0x9F, 0xA9, 0x40, 0xC9 };

/* ABP para*/
uint8_t NwkSKey[] = { 0xAF, 0x71, 0x63, 0x3F, 0xF3, 0xB4, 0xD4, 0x33, 0xE4, 0xED, 0x43, 0x1F, 0x87, 0x98, 0xAD, 0xF8 };
uint8_t AppSKey[] = { 0xB3, 0x68, 0x0C, 0x15, 0xCC, 0x1B, 0x9B, 0xD1, 0xB7, 0x0D, 0x4D, 0x04, 0x67, 0xAB, 0x45, 0xF4 };
uint32_t DevAddr =  ( uint32_t )0x26012F35;

/*LoraWan Class, Class A and Class C are supported*/
DeviceClass_t  loraWanClass = CLASS_A;

/*the application data transmission duty cycle.  value in [ms].*/
uint32_t appTxDutyCycle = 15000;

/*OTAA or ABP*/
bool overTheAirActivation = true;

/*ADR enable*/
bool loraWanAdr = true;

/* Indicates if the node is sending confirmed or unconfirmed messages */
bool isTxConfirmed = true;

/* Application port */
uint8_t appPort = 2;

/*!
* Number of trials to transmit the frame, if the LoRaMAC layer did not
* receive an acknowledgment. The MAC performs a datarate adaptation,
* according to the LoRaWAN Specification V1.0.2, chapter 18.4, according
* to the following table:
*
* Transmission nb | Data Rate
* ----------------|-----------
* 1 (first)       | DR
* 2               | DR
* 3               | max(DR-1,0)
* 4               | max(DR-1,0)
* 5               | max(DR-2,0)
* 6               | max(DR-2,0)
* 7               | max(DR-3,0)
* 8               | max(DR-3,0)
*
* Note, that if NbTrials is set to 1 or 2, the MAC will not decrease
* the datarate, in case the LoRaMAC layer did not receive an acknowledgment
*/
uint8_t confirmedNbTrials = 8;

/*LoraWan debug level, select in arduino IDE tools.
* None : print basic info.
* Freq : print Tx and Rx freq, DR info.
* Freq && DIO : print Tx and Rx freq, DR, DIO0 interrupt and DIO1 interrupt info.
* Freq && DIO && PW: print Tx and Rx freq, DR, DIO0 interrupt, DIO1 interrupt, MCU sleep and MCU wake info.
*/
uint8_t debugLevel = LL_NONE;

/*LoraWan region, select in arduino IDE tools*/
LoRaMacRegion_t loraWanRegion = LORAMAC_REGION_EU868;

static void prepareTxFrame( uint8_t port )
{
    appDataSize = 4;//AppDataSize max value is 64
        int Str1 = (bmp.readTemperature());
    appData[0] = (char) (Str1);
    appData[1] = 0x01;
    appData[2] = 0x02;
    appData[3] = 0x03;
}

// Add your initialization code here
void setup()
{
   bmp.begin();
  if(mcuStarted==0)
  {
    LoRaWAN.displayMcuInit();
  }
  Serial.begin(115200);
  while (!Serial);
  SPI.begin(SCK,MISO,MOSI,SS);
  Mcu.init(SS,RST_LoRa,DIO0,DIO1,license);
  deviceState = DEVICE_STATE_INIT;
}

// The loop function is called in an endless loop
void loop()
{
  switch( deviceState )
  {
    case DEVICE_STATE_INIT:
    {
      LoRaWAN.init(loraWanClass,loraWanRegion);
      break;
    }
    case DEVICE_STATE_JOIN:
    {
      LoRaWAN.displayJoining();
      LoRaWAN.join();
      break;
    }
    case DEVICE_STATE_SEND:
    {
      LoRaWAN.displaySending();
      prepareTxFrame( appPort );
      LoRaWAN.send(loraWanClass);
      deviceState = DEVICE_STATE_CYCLE;
      break;
    }
    case DEVICE_STATE_CYCLE:
    {
      // Schedule next packet transmission
      txDutyCycleTime = appTxDutyCycle + randr( -APP_TX_DUTYCYCLE_RND, APP_TX_DUTYCYCLE_RND );
      LoRaWAN.cycle(txDutyCycleTime);
      deviceState = DEVICE_STATE_SLEEP;
      break;
    }
    case DEVICE_STATE_SLEEP:
    {
      LoRaWAN.displayAck();
      LoRaWAN.sleep(loraWanClass,debugLevel);
      break;
    }
    default:
    {
      deviceState = DEVICE_STATE_INIT;
      break;
    }
  }
}
```

Heltec-Aaron-Lee, 2020-04-09:
Did you config a gateway already connected to the TTN?

Then, what's the problem now?

jaslo, 2020-11-01:
For TTN in the US, subband 1 is used. So would the userChannelsMask need to be changed?

G-3r4-d, 2021-09-25:
Does this code work with TTN v3 via ABP? I have some issues to connect my esp32 to TTN, I think is due to a new payload format on the server