HelTecAutomation/ESP32_LoRaWAN #44: TTN Credential Mods to Commissioning.h and LoRaMac-definitions.h

slowtoaccept, 2020-04-29:
Is there another set of instructions on how to do this than  https://robotzero.one/heltec-lora32-lorawan-node/? 

Commissioning.h has 34 lines and I can't find LoRaMac-definitions.h contrary to the instructions.

kolawilk, 2020-06-03:
I was confused too on first sight, but you can change these settings in the Arduino IDE, see the Readme: https://github.com/HelTecAutomation/ESP32_LoRaWAN#how-to-use-this-library
All other LoRaWAN credentials are defined in the *.ino file, just see the examples.
But I have an issue to recognize the TTN downlinks, because the node will listen on the (wrong) standard Datarate: https://github.com/HelTecAutomation/ESP32_LoRaWAN/issues/45

vincegre, 2020-06-03:
May you please confirm how I should copy/paste values from TTN in the INO file ? Should I keep by default the msb display of ttn with device EUI starting by 0x00,0x28 and Application EUI by 0x70,0x83 ?

Thanks

kolawilk, 2020-06-03:
I used the msb version of the EUIs and it worked.