HelTecAutomation/ESP32_LoRaWAN #46: LORAWAN_DEBUG_LEVEL not declared in this scope while compiling !

vincegre, 2020-06-03:
Hi

I try to compile the OTAA_OLED sample code to test my Lora 32 module but when I compile I get these:

```
Compiling sketch...
/home/vincen/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_HELTEC_WIFI_LORA_32_V2 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"HELTEC_WIFI_LORA_32_V2\"" "-DARDUINO_VARIANT=\"heltec_wifi_lora_32_V2\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32 -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/variants/heltec_wifi_lora_32_V2 -I/home/vincen/Arduino/libraries/ESP32_LoRaWAN-master/src -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI/src -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire/src /tmp/arduino_build_257396/sketch/OTAA_OLED.ino.cpp -o /tmp/arduino_build_257396/sketch/OTAA_OLED.ino.cpp.o
OTAA_OLED:89:22: error: 'LoRaWAN_DEBUG_LEVEL' was not declared in this scope
 uint8_t debugLevel = LoRaWAN_DEBUG_LEVEL;
                      ^
OTAA_OLED:92:33: error: 'ACTIVE_REGION' was not declared in this scope
 LoRaMacRegion_t loraWanRegion = ACTIVE_REGION;
                                 ^
Using library ESP32_LoRaWAN-master in folder: /home/vincen/Arduino/libraries/ESP32_LoRaWAN-master 
Using library SPI at version 1.0 in folder: /home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI 
Using library Wire at version 1.0.1 in folder: /home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire 
exit status 1
'LoRaWAN_DEBUG_LEVEL' was not declared in this scope
Invalid version 'Internal RTC2.0' for library in: /home/vincen/Arduino/libraries/ESP32_LoRaWAN-master
Invalid version 'Internal RTC2.0' for library in: /home/vincen/Arduino/libraries/ESP32_LoRaWAN-master
```

If I understand well the error messages I'm supposed to setup these variables from the Tools Menu of Arduino IDE but I don't have any extra choices for that !! Am I missing something obvious ?

Thanks

Vincèn

kolawilk, 2020-06-03:
Did you install the [Heltec_ESP32 development framework](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series)?

xxxVincent-L, 2021-08-13:
Hello did you solve the problem? I'm having the same problem with the upload process and I've made the correct selection in the Tools, but it still reports an error

> > Did you install the [Heltec_ESP32 development framework](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series)?
> 
> I did but it fails on one of the step with something regarding Azure....
> 
> ```
> vincen@Kubuntu-portable-de-Vincen:~/Arduino/hardware/heltec/esp32$ git submodule update --init --recursive
> fatal: URL non trouvée pour le chemin de sous-module 'libraries/AzureIoT' dans .gitmodules
> vincen@Kubuntu-portable-de-Vincen:~/Arduino/hardware/heltec/esp32$
> ```

wn1980, 2022-06-03:
Follow this instructions to solve this problem: [https://heltec-automation-docs.readthedocs.io/en/latest/esp32/quick_start.html#via-arduino-board-manager](https://heltec-automation-docs.readthedocs.io/en/latest/esp32/quick_start.html#via-arduino-board-manager) .