HelTecAutomation/ESP32_LoRaWAN #51: OLED shows nothing on Wireless Stick with OTAA_OLED.ino

Waldmensch1, 2020-07-28:
ESP32_LoRaWAN/examples/OTAA_OLED/OTAA_OLED.ino

The Example compiles fine. LoRa Sending works like a charme and data arriving in TTN. But the OLED stays dark. I have to something special to get the OLED run on wireless Stick.

 Do I have to modify somewhat in the libraries?  The OLED cannot be broken because it has shown something with the code from factory when I plugged in on USB first time. 

- Tested the actual code from Master branch
- The Heltec ESP32 Dev Boards library v1.1.0 is loaded in Arduino 
- The Heltec ESP32 Dev Boards Board 0.0.5 is loaded in Arduino Boardmanager
- The Board "Wireless Stick" is selected in Arduino Board selector.
- License code is pasted into code and all TTN IDs are set (TTN communication runs fine)
- serial output looks good:

15:30:56.554 -> confirmed uplink sending ...
15:30:57.648 -> receive data: rssi = -55, snr = 26, datarate = 5
15:31:12.465 -> confirmed uplink sending ...
15:31:13.562 -> receive data: rssi = -57, snr = 25, datarate = 5
15:31:28.396 -> confirmed uplink sending ...
15:31:29.493 -> receive data: rssi = -58, snr = 25, datarate = 5
15:31:44.489 -> confirmed uplink sending ...
15:31:45.578 -> receive data: rssi = -57, snr = 26, datarate = 5

Waldmensch1, 2020-07-28:
I put some code in to show the pins. These looks okay and match with WirelessStick mapping here: https://resource.heltec.cn/download/Wireless_Stick/Wireless_Stick.pdf

So no clue why nothing is shown on Display. Only serial output works and the whole LoRa works as well.

    case DEVICE_STATE_SEND:
    {
      LoRaWAN.displaySending();
      prepareTxFrame( appPort );
      LoRaWAN.send(loraWanClass);
      deviceState = DEVICE_STATE_CYCLE;
      Serial.print("SDA_OLED: ");
      Serial.print(SDA_OLED);
      Serial.print(" SCL_OLED: ");
      Serial.print(SCL_OLED);
      Serial.print(" RST_OLED: ");
      Serial.print(RST_OLED);
      Serial.print("\n");
      break;
    }

18:20:52.203 -> confirmed uplink sending ...
18:20:52.203 -> SDA_OLED: 4 SCL_OLED: 15 RST_OLED: 16
18:20:53.292 -> receive data: rssi = -60, snr = 28, datarate = 5
18:21:07.657 -> confirmed uplink sending ...
18:21:07.657 -> SDA_OLED: 4 SCL_OLED: 15 RST_OLED: 16
18:21:08.752 -> receive data: rssi = -60, snr = 22, datarate = 5

Waldmensch1, 2020-07-30:
I tested the "Wireless Stick FactoryTest" Example in the meantime, there the Display works like a charm. So it is not broken. This Example uses the heltec.cpp library

if I change on top of ESP32_LoraWAN.cpp the Geometry this way to GEOMETRY_64_32 the display at least some illumination but only pixel. So I believe the Display integration in ESP32_LoraWAN.cpp is wrong somehow

#if defined( WIFI_LoRa_32 ) || defined( WIFI_LoRa_32_V2 ) 
	SSD1306  Display(0x3c, SDA_OLED, SCL_OLED, RST_OLED);
#elif defined( Wireless_Stick )
	SSD1306  Display(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_64_32);
#endif

grichfitz, 2022-07-14:
I also had issues with the OLED on the wireless stick I solved this by doing the following:
  // Setup Display
  pinMode(16, OUTPUT);
  digitalWrite(16, LOW);
  delay(50);
  digitalWrite(16, HIGH);
  Display.init();
  Display.setFont(ArialMT_Plain_16);
  Display.flipScreenVertically();
  Display.setTextAlignment(TEXT_ALIGN_LEFT);

  Display.clear();
  Display.drawString(1, 0,  "Hello World");
  Display.display();