HelTecAutomation/ESP32_LoRaWAN #52: Reset loop / panic on Joining....

Waldmensch1, 2020-08-26:
Platformio.ini:
```
[env:heltec_wireless_stick]
platform = espressif32@1.9.0
board = heltec_wireless_stick
framework = arduino
upload_port = COM6
monitor_speed = 115200
lib_deps = 
    https://github.com/mikalhart/TinyGPSPlus
    https://github.com/HelTecAutomation/Heltec_ESP32
    https://github.com/HelTecAutomation/ESP32_LoRaWAN
```

Log of build and Monitor of one OTAA examples. I inserted some outputs leeding with "next" to show the steps and used pins

```
> Executing task in folder WirelessStickGPS: C:\Users\tommy\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor <

Processing heltec_wireless_stick (platform: espressif32@1.9.0; board: heltec_wireless_stick; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wireless_stick.html
PLATFORM: Espressif 32 1.9.0 > Heltec Wireless Stick
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.10002.190628 (1.0.2)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TinyGPSPlus> 1.0.2 #cfb60e5
|-- <Heltec_ESP32> 1.1.0 #c8ddaff
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <ESP32_LoRaWAN> Internal RTC2.0 #ea67dc9
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\heltec_wireless_stick\src\main.cpp.o
Compiling .pio\build\heltec_wireless_stick\libd28\Heltec_ESP32\BMP180.cpp.o
Compiling .pio\build\heltec_wireless_stick\libd28\Heltec_ESP32\heltec.cpp.o
Compiling .pio\build\heltec_wireless_stick\libd28\Heltec_ESP32\lora\LoRa.cpp.o
Compiling .pio\build\heltec_wireless_stick\libd28\Heltec_ESP32\oled\OLEDDisplay.cpp.o
Compiling .pio\build\heltec_wireless_stick\libd28\Heltec_ESP32\oled\OLEDDisplayUi.cpp.o
Linking .pio\build\heltec_wireless_stick\firmware.elf
Retrieving maximum program size .pio\build\heltec_wireless_stick\firmware.elf
Checking size .pio\build\heltec_wireless_stick\firmware.elf
Building .pio\build\heltec_wireless_stick\firmware.bin
esptool.py v2.6
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   5.0% (used 16364 bytes from 327680 bytes)
Flash: [=         ]   7.9% (used 265688 bytes from 3342336 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM6
Uploading .pio\build\heltec_wireless_stick\firmware.bin
esptool.py v2.6
Serial port COM6
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:88:14:0c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash params set to 0x0230
Compressed 16848 bytes to 10928...
Wrote 16848 bytes (10928 compressed) at 0x00001000 in 0.1 seconds (effective 949.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1890.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 6553.6 kbit/s)...
Hash of data verified.
Compressed 267264 bytes to 141466...
Wrote 267264 bytes (141466 compressed) at 0x00010000 in 2.2 seconds (effective 971.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================================================================================= [SUCCESS] Took 22.67 seconds =========================================================================================
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
next SPI.begin with SCK=5, MISO=19, MOSI=27, SS=18
next Mcu.init with SS=18, RST_LoRa=14, DIO0=26, DIO1=35
ESP32 MCU init...
ESP32 MCU inited OK!
next LoRaWAN.init with loraWanClass=2,loraWanRegion=5

LoRaWAN EU868 Class C start!

next LoRaWAN.join()
joining...
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d40fb  PS      : 0x00060330  A0      : 0x800d3e8e  A1      : 0x3ffb1ef0
A2      : 0x3ffb1f24  A3      : 0x3ffb1f1c  A4      : 0x00000031  A5      : 0x3f400fd9
A6      : 0x3ffb1ea0  A7      : 0x00000008  A8      : 0x00000000  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x3ffc19c4  A12     : 0x3ffc0008  A13     : 0x00000000
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400d40fb:0x3ffb1ef0 0x400d3e8b:0x3ffb1f10 0x400d17e7:0x3ffb1f50 0x400d127a:0x3ffb1f90 0x400d8cf1:0x3ffb1fb0 0x400888fd:0x3ffb1fd0

Rebooting...
```

TwentySixer, 2020-08-26:
Looks like a similiar issue like https://github.com/HelTecAutomation/ESP32_LoRaWAN/issues/25

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32

SteveDSP, 2021-02-28:
I'm using Platform IO along with the Heltec ESP32 LoRa V2 board. I can compile, program and run the OTAA_OLED example. 

Board boots, joins but when LoRa.Sleep is called......

_Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled._

And the board reboots and the cycle continues....

Any ideas?

nzaharov, 2021-05-18:
Having the same problem with `.join()`. Error is traced to line 134 in LoRaMacConfirmQueue.c.
Only managed to use the library on v2.0.0 from 2 years ago...