HelTecAutomation/ESP32_LoRaWAN #57: Add LA915 frequency plan to LoRaMac.c for Brazil

pedrocavaglieri, 2020-12-25:
Hi @Heltec-Aaron-Lee @solotaker @gibus27 !

I have some issue when I set the ESP32 V2 to LoRaWAN network on BRAZIL, my region uses frequency plan (LA915-928A), so when I try to choose the correct LoRaWAN Region on Arduino IDE`s tools, I can't choose the suitable frequency plan. Is it possible to add a frequency plan to LoRaMac.c file to add a new frequency plan? Thanks!

I found a Region LA915-928A config files on the zipped file on the EVERYNET`s link below, see if it can help:
https://ns.docs.everynet.io/channel_plans/Stackforce.html

Thanks a lot!

Heltec-Aaron-Lee, 2020-12-26:
We will look into that.

pedrocavaglieri, 2021-01-19:
Hi @Heltec-Aaron-Lee 

Have you made any progress in this topic? The product that my team and I are developing depends on it.

Thank you very much.

vinivinipaes, 2021-02-03:
Hi. I also have the same problem. It's not possible to choose the frequency plan for Brazil (LA915-928A).
If you can add this frequency plan, please... we'd appreciate a lot. Thank you!

Polkorny, 2021-08-13:
Hi. Have you made any progrss in this topic ?

Quency-D, 2021-08-19:
We have added the LA915 code for WiFi LoRa 32 (V2), but we do not have a related server. Can you please help us test it, or recommend a related server, shall we test it?

pedrocavaglieri, 2021-09-02:
Thank you very much for adding the LA915.
I will test it on the next days and then I give you the feedback.