HelTecAutomation/ESP32_LoRaWAN #60: Proper settings for connecting to US 915MHz Helium LoRAWAN network?

gregarican, 2021-01-23:
I obtained my HelTec license key and have defined the proper NwkSKey, AppSKey, DevAddr, DevEui, AppEui, and AppKey. But my ESP32 V2 LoRaWAN board still isn't joining the Helium network. I am in the US, so here are the settings I am using:

`/*LoraWan channelsmask, default channels 0-7*/
uint16_t userChannelsMask[6]={ 0x00FF,0x0000,0x0000,0x0000,0x0000,0x0000 };

/*LoraWan Class, Class A and Class C are supported*/
DeviceClass_t  loraWanClass = CLASS_C;`

I've defined my board as REGION_US915 in the Arduino settings. And in the ESP_LoRaWAN.cpp file I modified this line to define the data rate:

`/*!
 * Default datarate
 */
#define LORAWAN_DEFAULT_DATARATE                    DR_3`

What else could I be missing for properly joining the network for a US board?

gregarican, 2021-01-23:
Modified the LoRaWAN class and channels to below. Still not joining into the Helium network.

`/*LoraWan channelsmask, default channels 0-7*/
uint16_t userChannelsMask[6]={ 0xFF00,0x0000,0x0000,0x0000,0x0000,0x0000 };

/*LoraWan Class, Class A and Class C are supported*/
DeviceClass_t  loraWanClass = CLASS_A;`

gregarican, 2021-01-23:
The most recent settings worked fine for US 915 MHz Helium networking. Was too far from LoRaWAN peer and once within range packets were successfully sent/received.

closed 2021-01-23 by gregarican

gfcbr, 2022-02-23:
Did you make it work?