HelTecAutomation/ESP32_LoRaWAN #62: join failed, rejoin at 30000 ms later / DIO1:RX Timeout

Polkorny, 2021-05-21:
join failed, rejoin at 30000 ms later

I'm using a Heltec WiFi 32 LoRa V2. I tried to use it on 2 different gateways, but the same error occured, the gateway receives the TX, but the node doesn't receive the RX

12:59:53.617 -> ESP32 MCU inited OK!
12:59:54.459 -> 
12:59:54.459 -> LoRaWAN AU915 Class A start!
12:59:54.459 -> 
12:59:54.459 -> joining...
12:59:54.459 -> TX on freq 915200000 Hz at DR 4
12:59:54.600 -> DIO0:TX Done
12:59:59.608 -> RX on freq 923300000 Hz at DR 12
12:59:59.608 -> DIO1:RX Timeout
13:00:00.637 -> RX on freq 923300000 Hz at DR 8
13:00:00.684 -> DIO1:RX Timeout
13:00:01.526 -> TX on freq 916600000 Hz at DR 0
13:00:03.258 -> DIO0:TX Done
13:00:08.266 -> RX on freq 927500000 Hz at DR 8
13:00:08.313 -> DIO1:RX Timeout
13:00:09.249 -> RX on freq 923300000 Hz at DR 8
13:00:09.389 -> DIO1:RX Timeout
13:00:09.529 -> join failed, rejoin at 30000 ms later

Polkorny, 2021-05-21:
I solved changing the mode LoRa public to private

closed 2021-05-21 by Polkorny

Jaedotmoe, 2021-11-02:
@Polkorny care to explain how to do that, please? I've been stuck with the exact same problem.

dtylerlivingston, 2021-11-29:
I'm stuck in same situation.  Not sure how to change to private 

joining...
TX on freq 903000000 Hz at DR 4
DIO0:TX Done
RX on freq 923300000 Hz at DR 13
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
TX on freq 904900000 Hz at DR 0
DIO0:TX Done
RX on freq 926300000 Hz at DR 10
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
join failed, rejoin at 30000 ms later

italocjs, 2021-12-02:
Impossible to make it work lol, anyone has any idea?  Gateway is working fine at ~5m from me.

dtylerlivingston, 2021-12-03:
The network started working for me this afternoon after failing to fully connect for 3 days. I changed nothing.  I read another post where others mentioned similar, it appears to have been network issue.

Jaedotmoe, 2021-12-03:
Still the same here, even though the device is near a gateway.

italocjs, 2021-12-03:
please see this: https://www.thethingsnetwork.org/forum/t/unable-to-join-ttn-gateway-receive-the-message/53625

I was trying to use a single channel gateway, which is not supported by ttn.

ChaiFox, 2023-04-18:
> I solved changing the mode LoRa public to private

I know this is closed but in case anyone else has the issue (like I did) and finds this thread LORAWAN_PUBLIC_NETWORK is located in the Commissioning.h file.