HelTecAutomation/ESP32_LoRaWAN #82: Building example sketches failed (PlatformIO)

ade21, 2022-06-10:
Hi there,

I am trying to build this example in a new PlatformIO project for my Hellten Wireless Stick:
https://github.com/HelTecAutomation/ESP32_LoRaWAN/blob/master/examples/OTAA/OTAA.ino

I am getting the following error:
[image]

Seems like this is a known issue in the Heltec forums:
http://community.heltec.cn/t/error-compiling-example-esp32-v2/10327

Can anybody help me out here?

Cheers
Andre

ameerhazo, 2022-06-22:
Hi @ade21 I'm facing the same problem trying to build the sketch. Did you get it to work on your end?

Regards,
Ameer

fabiano182, 2022-06-24:
Hi @maxgerhardt,

I was facing the same errors with rtc-board.c, your PR fixed them. But I'm with similar problems. Following pic:

[image]

I'm trying to use FreeRTOS with core 0 (BLE Task) and core 1 (LoRa OTAA Task). Would you know how to help me?

Best regards,
Fabiano

maxgerhardt, 2022-06-24:
I think that's a loosing game, depending on what code you have compiled it will shift the object files in the memory space and then suddenly other `call8` instructions have to be fixed than what's shown in your current error..

I've updated my fork with a few more fixes, so if you delete the `.pio` folder and build again it should hopefully show less errors. Try fixing the rest in the same way I did in the PR.

Heltec just needs to republish this library but compiled with `-mlongcalls` as [they promised](http://community.heltec.cn/t/error-compiling-example-esp32-v2/10327/6) but not yet delivered.

fabiano182, 2022-06-28:
Thank you @maxgerhardt 

Almost all errors with call8 gone away. My code still don't work but i think thats not library/Platformio problems.

Best regards,
Fabiano