HelTecAutomation/ESP32_LoRaWAN #87: LORAWAN_PREAMBLE_LENGTH doesn't found

oscardimanno, 2022-08-17:
I'm trying to run default examples without changing nothing, but i've got some errors.

How i can resolve?
Thanks

```
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionCN470.c: In function 'RegionCN470TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionCN470.c:550:64: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
     Radio.SetTxConfig( MODEM_LORA, phyTxPower, 0, 0, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3000 );
                                                                ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionCN470.c:550:64: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionAU915.c: In function 'RegionAU915TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionAU915.c:592:72: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
     Radio.SetTxConfig( MODEM_LORA, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
                                                                        ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionAU915.c:592:72: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionEU433.c: In function 'RegionEU433TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionEU433.c:634:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
                                                                       ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionEU433.c:634:71: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionAS923.c: In function 'RegionAS923TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionAS923.c:693:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
                                                                       ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionAS923.c:693:71: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionCN779.c: In function 'RegionCN779TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionCN779.c:634:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
                                                                       ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionCN779.c:634:71: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionEU868.c: In function 'RegionEU868TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionEU868.c:664:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3000 );
                                                                       ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionEU868.c:664:71: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionIN865.c: In function 'RegionIN865TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionIN865.c:638:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
                                                                       ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionIN865.c:638:71: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionKR920.c: In function 'RegionKR920TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionKR920.c:635:72: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
     Radio.SetTxConfig( MODEM_LORA, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 4e3 );
                                                                        ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionKR920.c:635:72: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionLA915.c: In function 'RegionLA915TxConfig':
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionLA915.c:669:72: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
     Radio.SetTxConfig( MODEM_LORA, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 4000 );
                                                                        ^
C:\Users\o.dimanno\Documents\Arduino\libraries\ESP32_LoRaWAN-master\src\region\RegionLA915.c:669:72: note: each undeclared identifier is reported only once for each function it appears in
exit status 1
Errore durante la compilazione per la scheda WiFi LoRa 32(V2).

```

moncada07, 2022-08-17:
Hey there! I had the same issue right now, I was able to solve it in the following way. Go to your Arduino libraries and search ESP32_LoRaWAN-master/src/region. Then, open all the region.c files, for example, RegionAS923.c and look for LORAWAN_PREAMBLE_LENGTH in code and just change it for a 16 and save all the files with same changes,  run the code again and that's all. But, after that if you're using TTN console, you will see another issue which is TTN console doesn't show the uplinks, I guess it's a kind of bug. I opened a issue some minutes ago. I will wait for answer or if you don't have this issue, let me know please. Good luck. @oscardimanno

moncada07, 2022-08-17:
Another thing is, remember to change the keys (DevEUI, APPKey, APPEUI) for yours. You can get them after to add a new end device on TTN Console (Your board is your end device) @oscardimanno

urs8000, 2022-08-23:
thanks for that work around ...
why couldn't that be declared globally when it's the same for all regions or could they be different?

urs8000, 2022-08-23:
@moncada07 I was too enthusiastic ;-)
I changes all regions and it (example OTAA_OLED for Wireless Stick) compiled successfully.
Running gives en endless loop of:
`18:14:08.922 -> ELF file SHA256: 0000000000000000
18:14:08.922 -> Rebooting...
18:14:09.070 -> assert failed: xQueueGenericSend queue.c:820 (pxQueue)
18:14:09.070 -> Backtrace:0x40083b21:0x3ffb25800x40089129:0x3ffb25a0 0x4008ded5:0x3ffb25c0 0x40089922:0x3ffb26f0 0x400d9b42:0x3ffb2730 0x400d9b89:0x3ffb2750 0x400d72eb:0x3ffb2770 0x400fa585:0x3ffb2790 0x400d7e25:0x3ffb27b0 0x400d186e:0x3ffb27f0 0x400dc062:0x3ffb2820 
`
any idea what's wrong?

moncada07, 2022-08-23:
> I guess that's because each region is different, and you have to declare
> one by one. I'm not sure about it.

I invite you to visit our repo GitHub and check it with the README to
compare and make the correct validations for your projects, also you can
see other modules for Wireless Stick Lite, using arduino. Actually I am
sure this will help you to resolve your problem and if you have any
trouble you can open an issue and I will reply to you, right now I am doing
a maintenance to complete all the repo, but OTAA module is ready. Go ahead!

https://gitlab.com/lantern-public/domus-lantern

moncada07, 2022-08-24:
Hey there, how are you? I want to ask you, did you visit our repository to
find out the documentation you were looking for? If yes, what did you think
about it, yesterday at night I made one of the last commits to finish it
around today's afternoon. I appreciate your feedback. See you.

urs8000, 2022-08-25:
@moncada07 and otaa_latern does not even compile.
`D:\_development\Arduino\libraries\arduino-lmic\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
D:\_development\Arduino\libraries\arduino-lmic\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
                                                                        ^
c:/users/ursm/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\ursm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32/lib\libpp.a(hal_mac.o): in function `hal_init':
(.text.hal_init+0xb4): multiple definition of `hal_init'; libraries\arduino-lmic\hal\hal.cpp.o:D:\_development\Arduino\libraries\arduino-lmic\src\hal/hal.cpp:416: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Heltec Wireless Stick.
`
it seems to be confused with installed libraries
I've installed "arduino-lmic" name=MCCI LoRaWAN LMIC library   version=4.1.1
and "arduino-lorawan" name=MCCI Arduino LoRaWAN Library  version=0.9.2

moncada07, 2022-08-25:
What kind of board are you using? Wireless stick lite?

urs8000, 2022-08-25:
@moncada07   no - a wireless stick
I've uncommented more in ReadChipID
// 21:15:07.449 -> ESP32ChipID=9xxxxxxxx4
// 21:15:07.449 -> ESP32 Chip model = ESP32-D0WDQ6 Rev 1
// 21:15:07.449 -> This chip has 2 cores

urs8000, 2022-08-25:
@moncada07 very strange! after doing it with ReadChipID I tried it again with your otaa_latern -- and it works.
the "unknown board" still appeard, but it seems to take the inbeded Pin mapping
why does this happen. I could only assume, that cleaning up while moving a CubeCell code to platformIO seems given that positive effect ;)
21:32:50.794 -> Starting
21:32:50.794 -> Packet queued
21:32:50.794 -> 315612: EV_JOINING
21:32:57.772 -> 751027: EV_TXSTART
21:33:02.880 -> 1072055: EV_JOINED
21:33:02.880 -> netid: 19
21:33:02.880 -> devaddr: 260BXXXX
21:33:02.880 -> AppSKey: B4-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-AF
21:33:02.920 -> NwkSKey: C8-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-69
21:33:02.920 -> 1072281: EV_TXSTART
21:33:08.068 -> 1395018: EV_TXCOMPLETE (includes waiting for RX windows)
21:33:09.553 -> 1489252: EV_TXSTART
21:33:15.674 -> 1871461: EV_TXCOMPLETE (includes waiting for RX windows)
21:34:15.676 -> 5621482: EV_TXSTART
21:34:15.676 -> Packet queued
21:34:21.798 -> 6004008: EV_TXCOMPLETE (includes waiting for RX windows)
21:35:21.799 -> 9754028: EV_TXSTART
21:35:21.799 -> Packet queued
21:35:27.940 -> 10136555: EV_TXCOMPLETE (includes waiting for RX windows)
---------------------------------------------------------------
and now also the example OTAA_OLED works
21:40:42.626 -> mcuStarted==0
21:40:43.107 -> MCUinit done
21:40:43.107 -> ESP32 MCU init...
21:40:43.147 -> ESP32 MCU inited OK!
21:40:48.826 -> 
21:40:48.826 -> DevEui=00Bxxxxxxxxxxxxx
21:40:48.867 -> LoRaWAN EU868 Class A start!
21:40:48.867 -> 
21:40:49.228 -> joining...
21:40:54.435 -> joined
21:40:55.840 -> unconfirmed uplink sending ...
21:41:01.018 -> receive data: rssi = -87, snr = 1, datarate = 5
21:41:21.435 -> unconfirmed uplink sending ...
for that I know I was replacing all Lorawan_PREAMBLE_Length in the regionxx as you suggested with 16

moncada07, 2022-10-11:
I guess that's because each region is different, and you have to declare
one by one. I'm not sure about it.

moncada07, 2022-10-11:
That's great! So, I guess you can connect now to your gateway.

jcwren, 2022-10-16:
Changing all the occurrences of `LORAWAN_PREAMBLE_LENGTH` to 16 is incorrect.

LoRaWAN uses a preamble length of 8 in every region. A device transmitting a 16 symbol preamble will be heard by a device expecting 8 symbols, but not vice versa. If you change all those occurrences of `LORAWAN_PREAMBLE_LENGTH` to 16, the gateway will hear your device, but your device will never hear the gateway. 

There's also no need to edit each file. Just declare the define globally in your Makefile, or in the case of PlatformIO, add `-D LORAWAN_PREAMBLE_LENGTH=8` to the `build_flags`. 

You can reference issue #91 for some additional details.

369Martin369, 2022-10-23:
> 

cannot find any entry for LORAWAN_PREAMBLE_LENGTH ?!?

jcwren, 2022-10-23:
> > 
> 
> cannot find any entry for LORAWAN_PREAMBLE_LENGTH ?!?

Define it in your `platformio.ini` file as such:
```
build_flags =
  -D REGION_US915
  -D ACTIVE_REGION=LORAMAC_REGION_US915
  -D LORAWAN_DEBUG_LEVEL=0
  -D LORAWAN_PREAMBLE_LENGTH=8
```

369Martin369, 2022-10-23:
> Hey there! I had the same issue right now, I was able to solve it in the following way. Go to your Arduino libraries and search ESP32_LoRaWAN-master/src/region. Then, open all the region.c files, [...]

Thank you. this patch did work for me. I wonder if there is a better solution than patching 10 files.....